Flipper-XFW/Xtreme-Firmware #357: QFlipper no longer sees device

Kikkawaryu, 2023-08-14:
### Describe the bug.

Updated to the latest XFW-0050_27072023 firmware, using the webupdate and tried the .tgz 

Update completes with no issues, its detected in browser as COM3 and in windows, qflipper states Permission denied, device is busy

Tried another windows 11 machine and states same error

Factory reset the flipper, updated same error
Tried another firmware, qflipper finds the device straight away

Tried other COM Ports, disabled bluetooth on flipper 
Tried older versions of firmware, was working previously on XFW 

### Reproduction

Update to XFW-0050_27072023 

Open qflipper

bip3r, 2023-08-22:
I had the same issue. It's the problem with qFlipper. I came back to earlier vetsion and it workshop fine with nwwest Xtreme.

Kikkawaryu, 2023-08-22:
> I had the same issue. It's the problem with qFlipper. I came back to earlier vetsion and it workshop fine with nwwest Xtreme.

Tried latest version down to 1.0.0 and everything in between, 1.0.0 says device unrecognised and stated to "clean" in recovery mode 
the rest said permission denied

bip3r, 2023-08-22:
I have qFlipper 1.3.0.
It worked with 0.47 uptill cirrent 0.50.

Kikkawaryu, 2023-08-22:
Ill give the App Version a try but its mainly transferring USB Scripts from PC to Flipper rather than taking SD Card each time

Ill see if it will connect to the mobile app thanks!

ClaraCrazy, 2023-08-29:
works fine for everyone else, not reproducable, no info to work with, therefore closed.

closed 2023-08-29 by ClaraCrazy