Flipper-XFW/Xtreme-Firmware #360: Opening subghz file from favorites plays the file repeatedly automatically

PowerUser64, 2023-08-19:
### Describe the bug.

When opening a subghz file from the favorites menu, it is automatically played. Not only that, the file is repeated until the application is exited, which oddly enough can be done by pressing the center button (normally this button would play the file).

### Reproduction

1. Add a subghz file to favorites
2. Open the file from the favorites menu
     - it will play automatically, but it shouldn't
4. Press the center button
    - this exits but it shouldn't

### Anything else?

This was introduced as a regression in the latest release.

Willy-JL, 2023-08-19:
It's not a big. YOU enabled "favorite timeout". This is exactly what it does. Runs the file for the amount of seconds you set when you run the file from favorites.

closed 2023-08-19 by Willy-JL

PowerUser64, 2023-08-19:
Oh! my bad. I didn't know that's what that setting did! Thanks for letting me know!