Flipper-XFW/Xtreme-Firmware #362: Bluetooth keyboard should forward ASCII input

tauoverpi, 2023-08-21:
### Describe the enhancement you're suggesting.

Bluetooth keyboard remote should pass the ASCII range directly rather than ignoring it with `input keyboard` in use other than the blessed exit sequence (C-c).

### Anything else?

Usecase involves faster testing of automation scripts, using an external program to control input, convenience.

Maybe a C-a C-x or similar sequence to exit in addition to the above such that C-c can be sent. Not part of the proposed enhancement though.

Willy-JL, 2023-08-21:
I don't get what you mean...

tauoverpi, 2023-08-22:
In short, using the Bluetooth Remote's Keyboard mode while controlling the flipper via `input keyboard` over serial should translate ASCII sequences to key presses such that sending `A` results in the same action on the flipper (it behaves as if you navigated to the `A` key and hit `ok` to send).

In short, flipper as a bluetooth keyboard "adapter".

Willy-JL, 2023-08-23:
This is not really doable as things are right now. Unfortunately interfaces on flipper are purely graphical and mostly abstracted away from the rest of the system. The current implementation of `input keyboard` uses a very sketchy workaround, the cli service receives your keystrokes through the serial connection then gets a handle to the GUI service and crawls from there up by checking what the current on-screen view port and display type are (and with how flipper interfaces are designed, this is 5 layers of abstraction). If it detects that the topmost widget type is a text input then it will register your keystrokes and forward them to it. This is not really extensible as it is, and the other apps aren't really designed in a way that would make it easy to add support for this. The same workaround would not work here for example since the bt remote app is external. This would need a proper implementation where widgets and applications can subscribe to keystrokes and handle them on their own, separately from the GUI input pipeline.

Willy-JL, 2024-02-02:
Update, I implemented a modular API that allows any app or interface to subscribe to ascii input. Currently, the only source of ascii input is `input keyboard`. But this means that the app in question could be tweaked to support keyboard input from a pc