Flipper-XFW/Xtreme-Firmware #365: Can't open Hub in mobile app

vstkl, 2023-08-23:
### Describe the bug.

With official firmware it works as it should, but mobile app looks like it keeps refreshing for some reason, if someone has some pointers where to look I'd gladly try to fix the bug. 
I think the hub feature is quite new, I am not really following the updates and just play around with my flipper, as I said, advice appreciated and i will gladly fix it

### Reproduction

1. Open mobile Flipper App on iPhone
2. Click on hub
3.  Nothing appears

### Target

Mobile App

Willy-JL, 2023-08-23:
Duplicated of #347

closed 2023-08-23 by Willy-JL

vstkl, 2023-08-23:
thansk for clarification! @Willy-JL