Flipper-XFW/Xtreme-Firmware #368: Mass Storage

spyro12675, 2023-08-26:
### Describe the bug.

i have the new update on my flipper and when i click on mass storage it doesnt show anything
[image]
[image]
[image]

### Reproduction

1.USB
2.Mass Storage
3.Nothing

ClaraCrazy, 2023-08-26:
thats not a bug. it works by adding an img file to it.

ClaraCrazy, 2023-08-26:
@Willy-JL can explain things a lot better, so ig wait for him to wake up.

closed 2023-08-26 by ClaraCrazy

Willy-JL, 2023-08-26:
Yeah. Mass storage simply emulates a disk image. You need to give it a .img file and it will use that to emulate a usb drive. That .img file is essentially the raw data that you would have on the drive, not files and folders themselves, but the 1s and 0s that make them up. On the current v50 release, you need to make an image file yourself and upload it to flipper. On dev branch there is an option to create an empty disk image directly on flipper with the size you want.