Flipper-XFW/Xtreme-Firmware #370: BadKB: echo text on screen

Zarcolio, 2023-08-28:
### Describe the enhancement you're suggesting.

I've created several password and pin brute force scripts which work but there's no way to see which password or pin was typed and which one got accepted.
It would be nice if I could enable an option to echo the last typed text on the Flipper's screen, so I know which password/pin got accepted.

Willy-JL, 2023-12-10:
not so sure the whole text would fit. maybe a line counter would work better, and might be easier to implement as well.

Willy-JL, 2023-12-16:
now implemented via dae8a18bf724b2e7190a07dd9dec89317a9f79d2

shows current/total line number instead of the line text since that would not fit

closed 2024-02-02 by Willy-JL