Flipper-XFW/Xtreme-Firmware #372: Support for CAME 24bit (433.88 AM)

Gamper187, 2023-08-30:
### Description of the feature you're suggesting.

There are parking barriers that have a CAME 24bit protocol at 433.88 MHz. It would be nice if there is this function with more frequencies.

### Anything else?

It would be great if it supported the latest RFID transponder standards and NFC technologies.
Would it also be possible to add a bruteforcer for RFID and infrared because not every remote works 100%?

its0x08, 2023-10-06:
Totally agree!
LEGIC cards are used everywhere in EU and it feels like you have a useless device if those cards are not supported 😞.

Willy-JL, 2023-12-10:
CAME 24bit is already supported on 433.92, its close enough to 433.88 that it should not make any difference.

there is an argument to be had for LEGIC cards.

RFID bruteforce is already a thing. its called RFID fuzzer.

Infrared bruteforce is not a thing yet, theres already an issue open for it #225, although the current universal remote is already a sort of bruteforce, fuzzer would make more sense as a term since it means unknown codes.

Willy-JL, 2023-12-10:
could you provide any additional info or dumps for these cards/protocol?

spralexx, 2024-01-04:
@Willy-JL you can find more information including links to the proxmark implementation in this OFW Issue https://github.com/flipperdevices/flipperzero-firmware/issues/1688 

Slides from a talk at 26C3 including details on the protocol can be found here: https://fahrplan.events.ccc.de/congress/2009/Fahrplan/attachments/1506_legic-slides.pdf

jaessy77, 2024-02-22:
> could you provide any info or dumps for these cards/protocol?

PM me. Maybe I can provide useful dumps

jaessy77, 2024-02-22:
> could you provide any info or dumps for these cards/protocol?

PM me .. have a daily used Legic Card and Proxmark 3.

swissbyte, 2024-03-29:
Im also interrsted in getting legic work

hjakovski, 2024-06-14:
AFAIK there is no encryption at all. The data is just obfuscated and has a checksum to verify. 
Would be great to see an implementation on the Flipper.

parsley, 2024-07-03:
Any news about Legic on Flipper?