Flipper-XFW/Xtreme-Firmware #373: Gather Sub-GHz signals

Stepzor11, 2023-08-30:
### Description of the feature you're suggesting.

If you keep running the Sub-GHz read for a while, you might get a good number of records.

The idea is to add the ability to gather them and show only the different ones, so you wont miss a different signal in the middle of a bunch of repeated ones.

### Anything else?

Please let me know if this sounds like an interesting idea

htotoo, 2023-09-01:
Should be a setting in the subghz app, so if I want to see the repeated signal, I would, but while gathering, could filter the repeating ones.
+1

Willy-JL, 2023-12-10:
now implemented via 4eaa95999448287bcb28d0ddc2f6bfc90ec98161

it uses a hash of the signal, so there is a tiny chance that it might ignore non-duplicates.
also rolling code signals will repeat due to having different encrypted data and therefore hash.
this could be improved later on by making the hash function ignore dynamic/encrypted data.

Stepzor11, 2023-12-10:
> now implemented via 4eaa95999448287bcb28d0ddc2f6bfc90ec98161
>
> it uses a hash of the signal, so there is a tiny chance that it might ignore non-duplicates.
> also rolling code signals will repeat due [...]

Sounds pretty interesting, thanks!

If I can give you a suggestion, instead of ignoring, it would be better to just identity it as a duplicate; whether it is an icon, an incremental counter next to the first signal (x2, x3, ..., xN)  or with a specific button that postprocess all the read signals filtering the duplicates or toggling the counter mentioned before (this way you can hit "back" and see the full list and chronological order) (my favorite).

Pros of this approach:
- no signals are ignored
- you have a full view of what's happening in the sub-GHz world (it is useful to see new signals transmited, even if they are duplicates)

Stepzor11, 2023-12-10:
> one of my incentives implementing this ignore option is that ram is limited, so ignoring duplicates saves space for other signals to be recorded. but youre right that a counter could be interesting too. will probably make something basic like a counter on each signal, after receiving a new one it searches in the history for others with same hash, when found it will remove it if you have ignore enabled, otherwise it will add 1 to the latest counter for that hash and mark it on this signal. also means that while ignoring, you will see a number of how many times that signal was repeated

That'd be close to perfection 😁

Willy-JL, 2023-12-10:
now fully implemented via f96a6bde86ff438decad5a995e4a3e343ddc2b39...9fbb9ebbae08eab629c921f5dc6acf813293a275

also works when decoding a raw file, will respect the setting you choose

preview with remove duplicates
[image]

preview without remove duplicates
[image]

Willy-JL, 2023-12-10:
once again dynamic protocols dont work with this system yet but might eventually, just need a better hash implementation that ignores encrypted data and gives a meaningful hash

Stepzor11, 2023-12-10:
> now fully implemented via f96a6bde86ff438decad5a995e4a3e343ddc2b39...9fbb9ebbae08eab629c921f5dc6acf813293a275
> 
> also works when decoding a raw file, will respect the setting you choose
> 
> preview with remove duplicates
> [image]
> 
> preview without remove duplicates
> [image]
> 

Love it ❤️

Stepzor11, 2023-12-10:
> once again dynamic protocols dont work with this system yet but might eventually, just need a better hash implementation that ignores encrypted data and gives a meaningful hash

Of course 😁

Stepzor11, 2023-12-10:
Bonus idea: being able to switch on and off the duplicates counter after recording signals would be pretty interesting

Willy-JL, 2023-12-10:
it will go back through when a new signal is received if you toggle it on, will remove its previous duplicates. manually triggering it however not currently. we'll see

Stepzor11, 2023-12-10:
> it will go back through when a new signal is received if you toggle it on, will remove its previous duplicates. manually triggering it however not currently. we'll see
> 

I know 😁

But it would be quite sharp this way

Willy-JL, 2023-12-10:
all done!

Stepzor11, 2023-12-10:
Love ya❤️

closed 2024-02-02 by Willy-JL