Flipper-XFW/Xtreme-Firmware #375: LCD backlight problem

sariyan-0, 2023-09-02:
### Describe the bug.

I recently updated my flipper and now the lack light stays on even after the timeout and im keep restting my flupper and reinstalling the OS and nothing changes and I have even switched to the flippers original OS and the screen timeout works and it only happens on the new xtreme update.

### Reproduction

No idea!

Willy-JL, 2023-09-02:
disable the "rgb backlight". you do not have an rgb backlight. you have a regular backlight. when you enabled it, you were warned about this.

Willy-JL, 2023-09-02:
you can find it in Xtreme > Misc > Screen > RGB Backlight

closed 2023-09-02 by Willy-JL