Flipper-XFW/Xtreme-Firmware #376: Barcode light adjustment

Qreed213, 2023-09-02:
### Describe the enhancement you're suggesting.

I use a 5-10% brightness. 

when im when i use the barecodes i forguett that 5% light is to low to read so i need to go back, sett the brightnes to 100% then go to the correct barcode and scann.

can ther be a brigthness level like in the clock (( When pressing left buttorn ther is a clock and date with light )) 
so it dripps back to deffualt brigthnes aftera scann is done.
guess this it is good for some other apps to like QR but its barecode i use.

love the work that is done so far :D

MatthewKuKanich, 2023-10-11:
Just submitted a PR to the original repo adding a brightness change when viewing barcodes

Qreed213, 2023-10-11:
Im not so used to git, but i understand than it is the better aprotch to contact the orginal maker.

closed 2023-10-11 by ClaraCrazy

MatthewKuKanich, 2023-10-11:
> Im not so used to git, but i understand than it is the better aprotch to contact the orginal maker.

What I'm saying is that I submitted a fix to the original dev. If he deems it good, then he will merge it with his, and it will be pushed as the official release. Until then, you can compile the fixed fap from here: https://github.com/MatthewKuKanich/flipper-barcode-generator