Flipper-XFW/Xtreme-Firmware #378: Flipper Automations

N-coder82, 2023-09-05:
### Description of the feature you're suggesting.

Automations on flipper, triggers and actions such as:

Triggers:
- certain time
- certain subghz signal
- certain rfid/nfc detection
- certain battery level
- certain module connection
- certain app opening
Actions:
- send a signal (subghz, gpio, infrared, etc.)
- make a sound
- power off
- reboot
- etc.

Lastly, you can turn it into a platform that people can add triggers and actions.

### Anything else?

I just think this is a cool idea, it is not a big thing. This firmware is ___amazing!___ and I will continue using it, this is not a make or break thing, just something that would help me!

ClaraCrazy, 2023-09-05:
"its not a big thing" - Then go ahead and write it. Also, have fun running that on your phone bc the flipper isnt even remotely capable of doing any of that.

closed 2023-09-05 by ClaraCrazy

ClaraCrazy, 2023-09-05:
mh. maybe the flipper one can handle this once its out. who knows

N-coder82, 2023-09-05:
🤞🤞🤞