Flipper-XFW/Xtreme-Firmware #382: Restricted frequencies

Olum-OG, 2023-09-06:
Hello, I am quite new to the Flipper zero and it's functions. I live in Germany. I installed the Xtreme-Firmware. I did read a few Reddit post, saying that you shouldn't unlock restricted frequencies for legal reasons. But I also read, that the Xtreme-Firmware automatically unlocks all restricted Frequencies. Is that right? Is it possible to use this firmware without unlocking restricted frequencies? I don't want to get in trouble for breaking the law, so I do not want to unlock these frequencies. It would be awesome if anyone could help me with my questions.

Thank you

N-coder82, 2023-09-06:
Turn it off in:

Xtreme settings > protocols

ClaraCrazy, 2023-09-06:
you dont get in trouble for it. its just bullshit claims of official. Code is not illegal. Never has, never will be. I'm in germany as well.

Willy-JL, 2023-09-06:
> Turn it off in:
> 
> Xtreme settings > protocols

that is something else for the record

Olum-OG, 2023-09-06:
> you dont get in trouble for it. its just bullshit claims of official. Code is not illegal. Never has, never will be. I'm in germany as well.

So it's not traceable and can't get "triangulated" like some people say?

There is also another Question I have for a long time now and couldn't really find an answer to: Are the features the Flipper Zero have traceable? For example: If I open a gate with the sub gh function, could the owner of the gate (of course it's me) find out that it was a Flipper zero with serial number "XXX"? 
Or is it like completely anonymous?

Thank you very much for your quick reply by the way! It means a lot for me

Willy-JL, 2023-09-06:
if youre lucky your flipper subghz will reach a couple meters. what do you expect to triangulate with that 💀

Willy-JL, 2023-09-06:
and no, subghz is not traceable. its public radio waves. imagine you are shouting "the code for the gate is 1234 now open!". except it can only hear what is being said, not who said it. and anyone else can hear it too.

Olum-OG, 2023-09-06:
> if youre lucky your flipper subghz will reach a couple meters. what do you expect to triangulate with that 💀

This post just scared me a bit 😂
And I think you can expand the range with a external antenna, aren't you?

https://www.reddit.com/r/flipperzero/comments/11i06zp/why_you_shouldnt_unlock_restricted_frequencies/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button

Olum-OG, 2023-09-06:
> and no, subghz is not traceable. its public radio waves. imagine you are shouting "the code for the gate is 1234 now open!". except it can only hear what is being said, not who said it. and anyone else can hear it too.

Thanks you a lot, you really helped me understanding this better. 
Appreciate it :)

Willy-JL, 2023-09-06:
that reddit post is very goofy... yeah obviously if its an illegal band, devices in your country will not be using it because its illegal for them too. but in the same logic, if no other devices use it because its illegal, you will never receive a signal in that illegal band, so you have nothing to transmit on it. and on the flipside, we have legs, cars and planes. we can move between countries. and we can order tech from other countries. so why should it be restricted then?

Olum-OG, 2023-09-06:
> that reddit post is very goofy... yeah obviously if its an illegal band, devices in your country will not be using it because its illegal for them too. but in the same logic, if no other devices use [...]

Yeah, makes sense. Thank you!

closed 2023-09-06 by Olum-OG