Flipper-XFW/Xtreme-Firmware #384: Plugin: Apple BLE Spam

Ichtestenurmal, 2023-09-07:
### Describe the bug.

Hello folks,

the plugin Apple BLE Spam says when running:

 error:
"Load failed Bluetooth/apple—ble—spam.fap
Found unsatisfied imports"

How can I fix that?

### Reproduction

1. Fresh firmeware install via web installer
2. Start
3. Error

Willy-JL, 2023-09-07:
You fucked it up. Don't do things you don't understand. Install the latest dev build, don't try to Frankenstein a new app into an incompatible firmware.

closed 2023-09-07 by Willy-JL

Ichtestenurmal, 2023-09-07:
Where can I find it?

I have not added anything, the plugin came with the latest version that I have installed.

Willy-JL, 2023-09-07:
No, it's the latest discord dev build. Considering the timing of this issue, I believe I answered you in the discord, but let me know if it was someone else and you still need help

khmm12, 2023-09-07:
@Ichtestenurmal just use the recent nightly build. 

https://github.com/Flipper-XFW/Xtreme-Firmware/actions/workflows/build.yml

e.g. https://github.com/Flipper-XFW/Xtreme-Firmware/actions/runs/6104041853
[image]

0xspade, 2023-09-07:
@Willy-JL does the discord dev build requires some kind of subscription or it's free?

Willy-JL, 2023-09-07:
Free. Just join the discord and get the dev updates role by clicking on it. Or you can get the builds from GitHub actions too

0xspade, 2023-09-07:
done installing, now I'm loving it :)

Thanks @Willy-JL

khmm12, 2023-09-07:
@Willy-JL 

[off-topic]

is there any chance to get an invitation to the discord server? 🙏 I tried a weak ago, but was declined due to new accounts protection and still can't join. The nickname is the same as in GitHub.

Willy-JL, 2023-09-07:
Try now