Flipper-XFW/Xtreme-Firmware #385: banned on xtreme discord for no reason

unkn0wn3xpl01t, 2023-09-07:
### Describe the bug.

sorry to raise this as an issue, but i haven't found any other way to contact you admins. i wanted to join your discord for the dev version xtreme firmware, but when i verified my number, i was immediately banned by you. my discord account is not completely new, nor have i done anything negative. i wanted to ask if it is possible to unban me? dc username: unkn0wn3xpl01t

### Reproduction

n/a

### Target

n/a

### Logs

```Text
n/a
```

### Anything else?

n/a

Willy-JL, 2023-09-07:
the bot automatically banned you because your account was too new, at 41 minutes of age, and detected you as a bot

Willy-JL, 2023-09-07:
try now

unkn0wn3xpl01t, 2023-09-07:
okay, thank you willy for your quick response. it worked. :) thank you also for this nice kind of firmware. :)

closed 2023-09-07 by unkn0wn3xpl01t