Flipper-XFW/Xtreme-Firmware #387: [ESP32] WiFi Marauder app is missing in XFW-0051 01-09-2023

iftsv, 2023-09-08:
hello,
I flashed latest version of Xtreme-Firmware XFW-0051 01-09-203 and also flashed DevBoard according to this manual https://github.com/Flipper-XFW/Xtreme-Firmware/wiki/Generic-Guides
But when I open Apps section, I can't find [ESP32] WiFi Marauder app to work with flipper WiFi Devboard. Could you please help me? Thanks in advance

N-coder82, 2023-09-08:
It's in  new section called Wi-Fi, scroll ALL the way down

N-coder82, 2023-09-08:
Confused me also!

Willy-JL, 2023-09-08:
we really gotta remove those generic guides...

closed 2023-09-08 by Willy-JL

N-coder82, 2023-09-08:
> we really gotta remove those generic guides...

What do you mean?

Willy-JL, 2023-09-08:
that was some random bs that dyiing wrote on discord 7 months ago, then ported to the wiki and then stopped updating, now doesnt even have anythign to do with us anymore. and there are better guides elsewhere, there are just misleading since they are plain outdated

N-coder82, 2023-09-08:
Oh, ok, also wanted to ask: am I allowed to help with the wiki?

iftsv, 2023-09-08:
@N-coder82 @Willy-JL oh, really thanks for fast reply and your work, have a n1c3 day ^)