Flipper-XFW/Xtreme-Firmware #389: Bug with Discord Verify bot

ZekePolarisBSH, 2023-09-11:
### Describe the bug.

It keeps saying it is typed wrong but it is not. Am I going to get banned from the server?

### Reproduction

Joined
Verify
Bot fails test?

ClaraCrazy, 2023-09-11:
[image]

[image]

your stupidity is not a bug

closed 2023-09-11 by ClaraCrazy

ZekePolarisBSH, 2023-09-11:
Weird I typed it with F each time. Idk why the second one does not have F
in it. The first one was because my caps was already locked and I unlocked
it by accident. It isn't stupidity, it is called a mistake.

ZekePolarisBSH, 2023-09-11:
Also you removed something of yours's after a channel said something about it.

ClaraCrazy, 2023-09-11:
dafuq did I remove? nothing...

ClaraCrazy, 2023-09-11:
and yes... a "mistake"

*I typed the captcha wrong... must be a bug, lets waste their time on github instead of using my eyes and brain*

ZekePolarisBSH, 2023-09-11:
Calm down, it was a CFW with all animations and not a mistake with me, a
mistake with the keyboard. I have missing keys here.... Sas shared it
saying it was by you. And I can read it I was just wondering if one makes
too many wrong guesses would the bot ban them. That's all what I was
wondering. No need to get hostile for no reason.

ClaraCrazy, 2023-09-11:
We didnt remove anything. dunno what sas has to do with any of this either.

ZekePolarisBSH, 2023-09-11:
Are you saying you don't know Talking Sasquach? If you didn't removed
anything what happened to the CFW?...

ClaraCrazy, 2023-09-11:
bro speak english lol. Nothing got removed, nothing happened. what are you talking about? jesus fucking christ... no way in hell you're even 18+

ZekePolarisBSH, 2023-09-11:
I can't speak I can only type don't harass others because they are
disabled. Are you trolling me? You ignored the channel name after asking
about it. If I am not 18+ then you must be 9 because no one types like that
and talks down on other's English. For someone that is claiming that I am
not 18+ you sure need to grow up before talking. And yes something was
deleted. If not by you then by someone else that works on it. I never said
you directly but for you to get all defensive tells me otherwise. Specially
if you just going to think someone that is older than you can't be 18+. 🤣

ClaraCrazy, 2023-09-11:
"harass" 🤡

ClaraCrazy, 2023-09-11:
and since you clown are still too mentally slow and incompetent to just *fucking say* what was supposedly deleted, because none of us can read your useless mind... I cant help you. Have a great day, I'm praying that it rains brain today and you catch one