Flipper-XFW/Xtreme-Firmware #390: Merging update image into icon pack

M-T-1, 2023-09-11:
### Describe the enhancement you're suggesting.

First of all, thanks for all the work you (and all the other collaborators) do for this FW, it's awesome.

Suggestion:
The FW has options for the graphics pack, including SFW graphics packs.
The current "fw updated" page shows this image: [NSFW Update Image](https://github.com/Flipper-XFW/Xtreme-Firmware/blob/dev/assets/slideshow/update_default/frame_00.png)
Would it be feasible to make the update image dependent on which graphics pack is in use?
e.g. 
 - show a watchdogs themed update img if the selected graphics pack is watchdogs and an update image has been provided in the pack
  - show the (current, nsfw) xtreme themed update img if the selected graphics pack is nsfw and an update image has been provided in the pack
  - show a dolphin themed update img if the selected graphics pack is default (which would use whatever image the stock fw uses)
  ... etc
   
 Any thoughts?

Willy-JL, 2023-09-11:
That is part of the update package, not the firmware or on the flipper itself, can't feasibly be put into an asset pack unfortunately

closed 2023-09-11 by Willy-JL

SolarSciencePup, 2023-10-29:
Can there be a SFW release then?