Flipper-XFW/Xtreme-Firmware #393: Cant install apps from yhe flipper "app store"

devicemodder, 2023-09-16:
### Describe the bug.

I'd love to be able to install apps from the "app store" that's in the flipper mobile app while keeping xtreme. 

Is there any way to accomplish this?

### Reproduction

Open flipper phone app, go 
[image]
to Hub > Apps. "Not compatible with your firmware"

Willy-JL, 2023-09-16:
this is because OFW updated, and we haven't yet. wait for a new release, or use the latest dev build.

devicemodder, 2023-09-16:
Ah, didn't know that. I'll wait then. Thanks for letting me know.

dedsecproyt, 2023-09-20:
You can use my dev build from [My Workflow build](https://github.com/dedsecproyt/Flipper_Frimware/releases/tag/main)

Willy-JL, 2023-09-20:
No. You can use the official dev builds made by us, the xtreme team, in the official xtreme discord server, on in the official xtreme GitHub actions

dedsecproyt, 2023-09-20:
> No. You can use the official dev builds made by us, the xtreme team, in the official xtreme discord server, on in the official xtreme GitHub actions

But its work well, i use my build
PS. wery happy because I can make updates via GitHub +- 10 minutes goes to build

Willy-JL, 2023-09-20:
...
thats literally what we already made. months ago. shipped directly to discord. single click download. by someone reputable.

dedsecproyt, 2023-09-20:
> ... thats literally what we already made. months ago. shipped directly to discord. single click download. by someone reputable.

But you probably can't make custom builds, for example without games or extra plugins...

Willy-JL, 2023-09-20:
you can do what you want on your own. the code is open for that reason. but you are not gonna suggest your random repo builds as an alternative to the official ones

dedsecproyt, 2023-09-20:
> you can do what you want on your own. the code is open for that reason. but you are not gonna suggest your random repo builds as an alternative to the official ones

Good, but if anything, you can still use your own build as an alternative for now...

closed 2023-12-09 by Willy-JL