Flipper-XFW/Xtreme-Firmware #395: Automatic bleeding edge releases

molangning, 2023-09-19:
There is a automatic github actions that builds the firmware at https://github.com/Flipper-XFW/Xtreme-Firmware/actions/runs/6209401149 
From what I can see, all files are built in there. Is there a way to publish them in releases so that bleeding edge releases lovers get to experience the latest firmware?

molangning, 2023-09-19:
to clarify, I can download it directly from the builds but having it in releases will be nice as it is more convenient

ClaraCrazy, 2023-09-19:
its not a release tho, so its not in the release tab.

just use the discord bot that keeps sending the new one

closed 2023-09-19 by ClaraCrazy

flipperuser256, 2023-10-19:
Is there no other way than discord? Because I'm unable to join the discord... ;(

Willy-JL, 2023-10-19:
https://github.com/Flipper-XFW/Xtreme-Firmware/actions/workflows/build.yml

molangning, 2023-10-20:
> Is there no other way than discord? Because I'm unable to join the discord... ;(

You can write a cronjob that checks for updates then alert you

Thats what I did anyways