Flipper-XFW/Xtreme-Firmware #397: SD Card not reading

Lorenzo2013460, 2023-09-23:
### Describe the bug.

Okay, so the SD card, every time i put it in, when i start it up with it, it has the reading screen, stuck. After a bit, it says "Mounting SD Card failed". When i start it up without it, it will load up the menu, with the LED turning green, after a bit stopping, and green again, until the LED flashes red. The menu, has now only 3 apps: the settings, Xtreme settings and Apps.
Xtreme just says an error, Apps is just empty, but Settings works just fine.
I remember sometime when it happened the same, but when i reputted it in again, it worked. Now it doesn't anymore. Any help? 

### Reproduction

Putting the SD card before powering on:
1.Powering on
2.Stuck on the reading screen
3. After waiting, SD Card not mounter screen pops up

Not putting the sd card before powering on:
1.Powering on
2.Menu, with only 3 apps and sd card not mounted.

If i put it in:
1: LED flashes green, after a bit no more, and then green...
2: After failing, flashes red. 

### Target

SDCard

ClaraCrazy, 2023-09-23:
buy a proper sd card that supports spi

closed 2023-09-23 by ClaraCrazy

Lorenzo2013460, 2023-09-23:
> buy a proper sd card that supports spi

I have a Kingston sd card. Does it support SPI?

ClaraCrazy, 2023-09-23:
https://google.com

Lorenzo2013460, 2023-09-23:
> https://google.com

¯\_(ツ)_/¯

Lorenzo2013460, 2023-09-25:
> https://google.com

Seems like i will have to find out myself :(

jtbg, 2023-11-01:
@Lorenzo2013460 what specific model/specs is your Kingston microsd card?

Lorenzo2013460, 2023-11-01:
> @Lorenzo2013460 what specific model/specs is your Kingston microsd card?

Where do i see it?

jtbg, 2023-11-01:
Usually, the information is printed on the card itself. Besides the brand name Kingston, what else is written on the card?

Lorenzo2013460, 2024-06-06:
> Usually, the information is printed on the card itself. Besides the brand name Kingston, what else is written on the card?

Sry for answering for so long, on the front there is (under "Kingston") "SDCS/16GB" and under it "31662-003.A00LF" In the back, "CG 16G 1822" and under it "HVLKA86 109A 16" Maybe they mean **something?**