Flipper-XFW/Xtreme-Firmware #398: Rolling code add padors garage

SolarSciencePup, 2023-09-24:
### Description of the feature you're suggesting.

Can Padors garage doors be added to rolling codes?

SolarSciencePup, 2023-10-10:
Do I need to provide anything?

Willy-JL, 2023-10-31:
you didnt really provide anything to begin with, we cant come up with information out of thin air

you get back the same effort that you provide.

SolarSciencePup, 2023-11-06:
I haven't done this before so what do I need?

SolarSciencePup, 2024-01-09:
What do I need?

anickname028, 2024-02-11:
you need the EXACT model, and just so you know, using your flipper for this can make it to where your other keys don't work. they do NOT update with your flipper.

SolarSciencePup, 2024-03-24:
I have 3 remotes remote number is first number
1. wall mounted control
2&3. key fob
key fobs have 4 buttons
wall mounted has 3
button number is top down last number
1st button isn't recorded as it is hooked up to door

https://flpr.app/sf#path=subghz/1_2.sub&id=az8vhm&key=GE1zPZ7gyoeSN1ir75K1rQ

https://flpr.app/sf#path=subghz/1_3.sub&id=jYgKcy&key=0Q5-MENxu4mEalTKXwIRiQ

https://flpr.app/sf#path=subghz/2_2.sub&id=C1KeNE&key=T5PFsPersJ_hmozKR6LA0Q

https://flpr.app/sf#path=subghz/2_3.sub&id=qT2rbm&key=ATCigKu31_0tAXZH3BuKlA

https://flpr.app/sf#path=subghz/2_4.sub&id=uf52Ss&key=F1sIPgb28K5DRKfL-T1Zvg

https://flpr.app/sf#path=subghz/3_2.sub&id=rkAuqL&key=5Igd0_UTsnuQPQuBCBanTg

https://flpr.app/sf#path=subghz/3_3.sub&id=kWc9NC&key=cdMP6M805NE68_-r0_lGKg

https://flpr.app/sf#path=subghz/3_4.sub&id=tGbUw5&key=cFGv8wOpcYlsn1h8vcKfqw

SolarSciencePup, 2024-05-25:
Post if links need to be refreshed