Flipper-XFW/Xtreme-Firmware #400: Epson projector universal remote not working

daemoncat, 2023-09-28:
### Describe the bug.

Whenever i try to turn of an epson projector, it pops up the "Turn OFF?" dialog box around 60% but when i try to broadcast again the box is aready gone

### Reproduction

1. go to universal remote
2. find epson projector
3. go to projectors section
4. click turn off
5. wait
6. you have reproduced the bug

### Target

Universal Remote

N-coder82, 2023-09-28:
That is the problem of the person who made the IR remote, not the Xtreme team

daemoncat, 2023-09-28:
ok thx

closed 2023-09-28 by daemoncat

N-coder82, 2023-09-28:
No problem

JCoulter85, 2023-09-28:
Latest iOS update made some of the Bluetooth controller options not work

Willy-JL, 2023-09-28:
> Latest iOS update made some of the Bluetooth controller options not work

Make an appropriate issue, don't hijack unrelated issues.