Flipper-XFW/Xtreme-Firmware #401: iOS 17 update / Bluetooth

JCoulter85, 2023-09-29:
### Describe the bug.

When you open the YouTube & TikTok application, the pervious and next buttons do not work.

### Reproduction

Power on. 
center circle button
Applications
Bluetooth>Bluetooth Remote>TiTok/YT Shorts

JCoulter85, 2023-10-02:
Any update on this?

Flipdaworld, 2023-10-04:
I have the same issue and I really like that feature seem like nobody has the solution

Brereid98, 2023-10-07:
seems like it's due to an iOS change, the app uses mouse inputs to control TikTok which the phone doesn't seem to like, if you turn on Assistive Touch in the accessibility settings it allows mouse control on the iPhone and gets it working again

Flipdaworld, 2023-10-08:
@Brereid98 man.... thank you thank you thank you , Yu area legend brother  it's back working again

Brereid98, 2023-10-08:
You're welcome, happy to help

Willy-JL, 2023-12-09:
can you try on latest release v52?

Brereid98, 2023-12-10:
still doesn't seem to work unless assistive touch is on

Flipdaworld, 2023-12-10:
Yeah sucks Sent from my iPhone

anickname028, 2024-02-11:
it won't work without assistive touch, so just enable it. iOS doesn't have BT mouse support without Assistive Touch on.

Flipdaworld, 2024-02-11:
It did work without it at one point , when I first got my flipper I never used assistive touch to get that to work but now I have to use it for it to work