Flipper-XFW/Xtreme-Firmware #407: Please create telegram channel

d9j, 2023-10-04:
### Description of the feature you're suggesting.

Discord seems going to downhill  with recent bugs and "features" .
 telegram appears to be best alternative.

ClaraCrazy, 2023-10-04:
no

closed 2023-10-04 by ClaraCrazy

d9j, 2023-10-04:
@ClaraCrazy seriously ? one person decision maker.  will never donate a cent

ClaraCrazy, 2023-10-04:
oh, im sorry. wasnt aware you want @Willy-JL @Sil333033 and perhaps all community members to say no as well, hold on I'll do something rq :) 

also, dont care.

SkeletonMan03, 2023-10-04:
> ### Description of the feature you're suggesting.
> 
> Discord seems going to downhill with recent bugs and "features" . telegram appears to be best alternative.
> ### Anything else?
> 
> _No response_

No

finch-harold, 2023-10-04:
> ### Description of the feature you're suggesting.
> Discord seems going to downhill with recent bugs and "features" . telegram appears to be best alternative.
> 
> ### Anything else?
> _No response_

no

mcurranseijo, 2023-10-04:
nay

Brereid98, 2023-10-04:
> ### Description of the feature you're suggesting.
> 
> Discord seems going to downhill  with recent bugs and "features" .
> 
>  telegram appears to be best alternative.  
> 
> ### Anything else?
> 
> _No response_

No

d9j, 2023-10-04:
ok. sorry for  seems to be dumb suggestion. and thank you for responses.

N-coder82, 2023-10-04:
> ### Description of the feature you're suggesting.
> 
> Discord seems going to downhill  with recent bugs and "features" .
> 
>  telegram appears to be best alternative.  
> 
> ### Anything else?
> 
> _No response_

No

htotoo, 2023-10-04:
big no

d9j, 2023-10-04:
would anyone of you guys take time to explain  why no?

Sil333033, 2023-10-04:
no @d9j

Willy-JL, 2023-10-04:
no.
*refuses to elaborate*
[image]

d9j, 2023-10-04:
yes