Flipper-XFW/Xtreme-Firmware #412: NFC Emulation not working

Abbin44, 2023-10-08:
### Describe the bug.

I have read an NFC tag and acquired 32/32 keys, 16/16 sectors read, and saved it. However, it's not working when emulating the tag and trying to use it. The tag is a Mifare Classic 1K. 
There seems to be an issue from the official firmware that discusses the issue, which can be found [here](https://github.com/flipperdevices/flipperzero-firmware/issues/2577) 
I'm using the latest firmware XFW-0051_01092023

### Reproduction

1. Read tag
2. Save tag
3. Get missing keys if neccessary
4. Emulate saved tag

haroon-m, 2023-10-15:
I am facing the same bug. I have a proxmark3 and used it to clone a Mifare Classic 1K multiple times. The cards work perfectly fine. 

I was able to clone the original card with the flipper and also accuired 32/32 keys and 16/16 sectors read.  But the reader just doesnt accept the flipper zero when emulating the card.

I used the proxmark3 to read the cloned card and the flipper zero and dumped the content to a bin file to comapre.
The files are the same. I am guessing the actual data was cloned correctly by the flipper and that its the emulation part that is a problem.

N-coder82, 2023-10-15:
The hardware is the problem, flipper Devs said they couldn't find a good emulator chip for nfc