Flipper-XFW/Xtreme-Firmware #417: NFC crashes when reading tag

marciogranzotto, 2023-10-10:
### Describe the bug.

On `dev`(9ef4b15f683419afba98556e9ad2b9cd674abc5b), when reading an NFC device, in this cases a 
FreeStyle Libre CGM device, causes a crash.

The message after reboot is:
> Flipper crashed but has been rebooted
> MPU fault, possibly stackoverflow

This doesn't happen on the release version.

### Reproduction

1. Flash commit 9ef4b15f683419afba98556e9ad2b9cd674abc5b
2. Open NFC app from the app menu
3. Select `read`
4. Try to read an NFC tag/device
5. Crashes

### Target

f7-C

Sil333033, 2023-10-10:
What type NFC is used on your device? Can you send a result of the read (with release version ofc)

marciogranzotto, 2023-10-10:
Well... it looks like it's also crashing most of the time in the main branch.

I've tried to watch the logs while reading the tag:
```
>: log debug
Current log level: debug
Use <log ?> to list available log levels
Press CTRL+C to stop...
68397 [I][FuriHalNfc] Deinit OK
68527 [I][FuriHalNfc] Init OK
68532 [D][DolphinState] icounter 9978, butthurt 0
70710 [I][NfcWorker] NfcV detected
70875 [D][NfcV] Read SYSTEM INFORMATION...
70887 [D][NfcV]   UID:          E0 07 A0 00 0C 0E 27 79
70889 [D][NfcV]   DSFID 243, AFI 7, Blocks 207, Size 94, IC Ref 0
70891 [D][NfcV] Reading block 0/206
70902 [D][NfcV]   9D F4 88 1B
70904 [D][NfcV] Reading block 1/206
70914 [D][NfcV]   00 00 00 00
70915 [D][NfcV] Reading block 2/206
70926 [D][NfcV]   00 00 00 00
70928 [D][NfcV] Reading block 3/206
70938 [D][NfcV]   DD E4 06 05
70940 [D][NfcV] Reading block 4/206
70951 [D][NfcV]   1F 80 C8 03
70953 [D][NfcV] Reading block 5/206
70963 [D][NfcV]   C1 03 C8 B4
70964 [D][NfcV] Reading block 6/206
70974 [D][NfcV]   C8 58 1F 80
70976 [D][NfcV] Reading block 7/206
70987 [D][NfcV]   5F 00 BF 03
70989 [D][NfcV] Reading block 8/206
70999 [D][NfcV]   07 04 C8 CC
71000 [D][NfcV] Reading block 9/206
71011 [D][NfcV]   C8 D8 5E 00
71013 [D][NfcV] Reading block 10/206
71023 [D][NfcV]   1E 80 F3 03
71025 [D][NfcV] Reading block 11/206
71036 [D][NfcV]   ED 03 C8 E4
71038 [D][NfcV] Reading block 12/206
71048 [D][NfcV]   C8 14 5F 00
71051 [D][NfcV] Reading block 13/206
71061 [D][NfcV]   5F 00 DD 03
71062 [D][NfcV] Reading block 14/206
71072 [D][NfcV]   D8 03 C8 6C
71074 [D][NfcV] Reading block 15/206
71085 [D][NfcV]   C8 A0 1F 80
71087 [D][NfcV] Reading block 16/206
71097 [D][NfcV]   1E 80 8B 04
71100 [D][NfcV] Reading block 17/206
71110 [D][NfcV]   60 04 C8 8C
71111 [D][NfcV] Reading block 18/206
71121 [D][NfcV]   C8 E4 5E 00
71123 [D][NfcV] Reading block 19/206
71134 [D][NfcV]   1E 80 09 06
71136 [D][NfcV] Reading block 20/206
71146 [D][NfcV]   00 06 C8 B0
71149 [D][NfcV] Reading block 21/206
71159 [D][NfcV]   C8 4C 60 00
71160 [D][NfcV] Reading block 22/206
71170 [D][NfcV]   1F 80 D9 06
71172 [D][NfcV] Reading block 23/206
71183 [D][NfcV]   A5 08 C8 58
71185 [D][NfcV] Reading block 24/206
71195 [D][NfcV]   C8 90 1F 80
71198 [D][NfcV] Reading block 25/206
71208 [D][NfcV]   1F 80 5D 09
71209 [D][NfcV] Reading block 26/206
71219 [D][NfcV]   7C 09 C8 44
71221 [D][NfcV] Reading block 27/206
[... 224 lines cut ...]
Exception in thread rx:
Traceback (most recent call last):

--- exit ---
  File "/Users/marciorodrigues/Projects/Flipper/Xtreme-Firmware/toolchain/x86_64-darwin/python/lib/python3.11/site-packages/serial/serialposix.py", line 575, in read
    buf = os.read(self.fd, size - len(read))
```

If I try to read the tag via the CLI, I get:

```
>: nfc detect
Detecting nfc...
Press Ctrl+C to abort
Found: NFC-V UID length: 8, UID:79270E0C00A007E0
```

I've seen other crash messages though, like:
- NULL pointer dereference
- BusFault

Also, when it successfully reads the tag, it crashes when I press the "More" button, with `NULL pointer dereference` error.

But here's its screen when it successfully reads the tag:
[image]

Willy-JL, 2023-10-16:
can you try on latest dev branch?

also do you have asset packs enabled?

how are you launching the nfc app?

does it also happen just right after a reboot?

Willy-JL, 2023-12-09:
the nfc stack has been fully refactored. please check if it happens on the latest release

also ensure that settings > system > heap trace = none

Willy-JL, 2024-02-02:
Very stale, and there's been copious amounts of fixes for nfc and nfcv, I believe this is fixed. Report back if you still have issues

closed 2024-02-02 by Willy-JL