Flipper-XFW/Xtreme-Firmware #420: Ability to modify RGB LED color schemes

Sn0w3W, 2023-10-12:
### Describe the enhancement you're suggesting.

Hello!

Is it possible to implement an ability to change (add, remove, edit existing) color schemes for LCD LED with custom RGB values?

My screen become kinda blue-ish with White preset, so I want to calibrate it, add some new colors.

ClaraCrazy, 2023-10-12:
use the dev build from discord.

closed 2023-10-12 by ClaraCrazy

Sn0w3W, 2023-10-12:
@ClaraCrazy, I mean ability to modify these presets, not choosing them in Xtreme app.
[image]

I built and installed last dev. build form GH already, White preset is blue-ish compared to laptop screen:
[image]

ClaraCrazy, 2023-10-12:
you have the file open, why dont you modify it right there lol. Thats not doable to let users modify in code. Or just use your own custom preset like everyone else does and input hex. Also if that looks blue-ish to you then you might need glasses. Also also, you can literally see the rgb values for white. It doesnt get any whiter than that lol.

Sn0w3W, 2023-10-12:
kthxbye

ClaraCrazy, 2023-10-12:
🤨