Flipper-XFW/Xtreme-Firmware #426: Sub-GHz Radio Settings

bip3r, 2023-10-16:
### Describe the bug.

Within Sub-GHz, under Radio Settings it is not possible to choose 'External' for Module.
Only 'Internal' option available. Of course external module attached.
(this works in Unleashed firmware, but I would love to come back to Xtreme :) ).

### Reproduction

1. Attached ext. Module for Sub-GHz.
2. Sub-GHz menu.
3. Radio Settings
4. Module - Internal (External not available to choose)

### Target

Sub-GHz - Radio Settings - Module

### Anything else?

With unlished firmware it works ok.

Sil333033, 2023-10-16:
can you try this? 

Xtreme app -> protocols -> GPIO pins. set SPI CC1101 handle to default.

Willy-JL, 2023-10-16:
if only "internal" is available, either you have the wrong SPI handle selected or your module is not connected properly. if it works on unleashed, chances are you selected the extra SPI handle. do what sil said

Marckus95, 2024-05-04:
Hello, I have the same problem with Xtreme 0053, just internal. Thank you for your help.

sergio4d, 2024-05-07:
> Hello, I have the same problem with Xtreme 0053, just internal. Thank you for your help.

Same here! I cannot switch between internal and external subghz modules. Only internal is available. 
XFW 0053
SPI handle is default.

I tried giving it 5V, enabling subghz external in firmware settings > protocol, but that didn't help.

:'(

luciferseamus, 2024-06-24:
> Same here! I cannot switch between internal and external subghz modules. Only internal is available. XFW 0053 SPI handle is default.
>
> I tried giving it 5V, enabling subghz external in firmware [...]

I am having the same issue.  Funny note though: if I set the CC1101 SPI to extra then go into the SubG menu the external boards LED will light up but EXT module is not a selectable option. If however, I then go back to the SPI options and reset it to default and then return to the SubG menu EXT module becomes selectable.  However, the moment I try to use the antenna my Flipper will crash.