Flipper-XFW/Xtreme-Firmware #432: I would like to point something out

tkbstudios, 2023-10-24:
# Raiding RMs discord server is completely inappropriate and shouldn't be done, a new reason not to use your firmware..
## shame on you
[image]
[image]
[image]

Willy-JL, 2023-10-24:
huh???

this has nothing to do with us?

complain to the people doing it, which is not us from the xtreme team.

closed 2023-10-24 by Willy-JL

ClaraCrazy, 2023-10-24:
goofy ahh shit to wake up to...

tkbstudios, 2023-10-28:
Yeah you guys woke me up too with your aggressive pings from your bots