Flipper-XFW/Xtreme-Firmware #435: Unable to connect bluetooth on Win11

ashhad, 2023-11-01:
### Describe the bug.

I have just flashed the firmware and tried connecting the device to Bluetooth on my win11 machine. I get a prompt on the device with a PIN but when the same PIN is entered, I get an error (screenshot attached)

### Reproduction

1. Switch on Bluetooth
2. Find the device on the win11 machine
3. Click on Connect
4. Enter the PIN shown on FLipper device
5. Error message
6. 
[image]
[image]

Willy-JL, 2023-11-01:
You need to set "Bluetooth devices discovery" to advanced in settings for it to show

ashhad, 2023-11-01:
HI Willy,

It's already set to advance

[image]

Willy-JL, 2023-11-01:
Right... you are trying to pair the normal flipper connection. That doesn't work. You connect flipper to your phone, for the mobile app. Flipper doesn't work with pc over Bluetooth. You can connect it to pc Bluetooth if you are using badkb, bt remote, or similar things. But the normal flipper bt will not connect.

closed 2023-11-01 by Willy-JL