Flipper-XFW/Xtreme-Firmware #436: Genuinely can't decrypt the Discord Captcha

Kabouik, 2023-11-01:
I am sorry to open an issue for such a ridiculous issue, but I only have one attempt left on Discord, and if this is not `wdnP` or `wdmP`, then I really cannot tell what it is.

[image]

Maybe `wzdmP`? But with only one attempt left, I'm not willing to try before I can get sure my Discord account won't be blacklisted from the server and prevented from another join attempt.

Kabouik, 2023-11-01:
Apparently Discord expelled me before I could use my third attempt, and din't block me when trying the invitation link anew. That time, the captcha was clearer. Sorry for the noise.

A button to change captchas would be handy, but I suppose this is a Discord limitation.

closed 2023-11-01 by Kabouik

Willy-JL, 2023-11-01:
Failing the third time would've given a new captcha. It's 3 attempts per image, not before the end of the world