Flipper-XFW/Xtreme-Firmware #447: BadBT on macos

yevyevyev, 2023-11-10:
### Describe the bug.

Hello, first of all thank you for this firmware.
Does the badBT work on macos?

### Reproduction

1. Protocols -> Change from USB to BT
2. Starting BadBT for macos, it resets to USB
3. for every other platform (windows, ios, android) it works

N-coder82, 2023-11-10:
It works on any device that supports Bluetooth, however set the ID to a Apple keyboard ID to avoid a keyboard setup prompt showing up

Willy-JL, 2023-11-10:
Yeah if you try to change it you even get told "Script has ID cmd! Locked to USB mode!"
Make your own script, or remove the ID command from the script, or use any other script, and it works

closed 2023-11-10 by Willy-JL