Flipper-XFW/Xtreme-Firmware #450: After installing Xtreme-Firmware flipper zero cannot pair with any device

HeytalePazguato, 2023-11-13:
### Describe the bug.

After successfully installing Xtreme-Firmware it is not possible to pair using bluetooth with any device, I tried to check open and past issues and could not find any information about this. Is this normal?

### Reproduction

1. Install Xtreme-Firmware using the Web updater
2. Wait until the installation is done
3. Enable bluetooth connection in the Flipper Zero
4. Try to pair to any device

* Flipper zero is not even appearing when searching for bluetooth devices

### Target

iPhone 12, iPhone14, Windows 10

Willy-JL, 2023-11-13:
Unpair the devices both on flipper mobile app, on phone settings, and on flipper settings. This is caused by old connections made on ofw. Custom firmware has a different pairing method, old connections won't work and need to be reset.

closed 2023-11-13 by Willy-JL

HeytalePazguato, 2023-11-13:
No, this didn't resolve the issue.

In the Flipper Zero I forgot all paired devices, disable the Bluetooth , enable again, even rebooted it.
In the other devices I forgot the Flipper Zero and rebooted

Anyway it doesn't even appear while searching for devices

Why was the issue closed without confirming if the solution works?

Willy-JL, 2023-11-13:
> Why was the issue closed without confirming if the solution works?

Because we get this same question all the time. It's always the same issue and the same response.

In your case, it still is a very common issue I would have closed right away.

You cannot connect to flipper in normal settings. Use the flipper mobile app. It can't pair to pc either. This is for normal flipper Bluetooth, if you are doing badbt or bt remote those pair separately and work in normal settings

HeytalePazguato, 2023-11-13:
> Because we get this same question all the time. It's always the same issue and the same response.

I think then this should be addressed in the homepage.

Anyway, the solution you provided still doesn't work. I even deleted the FlipperZero app, re-installed and still the device can't be detected.

I formatted the SD card, reinstalled the original firmware and it works.
I installed the Xtreme firmware again and again it doesn't work.

I formatted the SD card again, reinstalled the original firmware, and without testing installed the Xtreme firmware, and then it worked.

It doesn't seem to be the same issue you described.

I assume the Flipper Zero "Forget all Paired devices" function is not really deleting the previous connections.

Willy-JL, 2023-11-13:
And it's addressed in the discord FAQ channel, and honestly it's basic troubleshooting steps to unpair and repair, so a normal user would figure it out on their own.

Diko-Jelev, 2024-01-17:
> And it's addressed in the discord FAQ channel, and honestly it's basic troubleshooting steps to unpair and repair, so a normal user would figure it out on their own.

Basically I have the same issue as the one above. I will play around a little bit. 

With me, it happened after I noticed that the device is remembered with the old BT device name so I decided to forget Bluetooth connections to reconnect. afterwards rep broadcasted to my iPhone (12 pro/ iOS 17.2.1). BadKB via bluetooth works just fine.

And yet the device is undiscoverable by the iPhone. XFW-0052

Calbrea, 2024-01-31:
Same problem as the OP. Windows 11 notebook, Windows 10 notebook and MacBook unable to find the flipper via bluetooth. I tried all of the troubleshooting steps from above.

kiyokodyele, 2024-02-06:
Same problem but only in windows. In android and iOS it works fine.

BigJohnTW69, 2024-02-09:
I had this very same issue and followed all the steps. What I figured out is that the Flipper mobile app stored information that confused the original device. In the mobile app you must disconnected (even though the flipper has been removed from the BT list) then forget this device in the flipper app. After that you will be able to connect to the flipper in the very same app... I hope it works for you as it has worked for me.

kjames2001, 2024-03-25:
same problem here, tried all possible step including flashing ofw.

RChadwick7, 2024-03-31:
I have a similar issue. When I go to pair, a 6 digit number usually pops up on the Flipper, as if I'm supposed to enter it in the phone. However, the number disappears within 1 seconds and the phone says failed.