Flipper-XFW/Xtreme-Firmware #458: Freeze While Emulating Amiibo

alphakdelta, 2023-11-20:
### Describe the bug.

While emulating amiibo files to be read by Switch, will randomly freeze and be stuck emulating the file. Flipper will not respond to anything but a hard reset. Happens on multiple different amiibo files. 

### Reproduction

1. Navigate to NFC 
2. Saved files 
3. Select Amiibo 
4. Amiibo collections
5. Choose any amiibo file within the subfolders 
6. Emulate 
7. Crash

### Logs

```Text
[19:14:58][info]: disconnected
[19:14:58][debug]: reconnecting
[19:14:58][info]: connecting
[19:14:58][info]: session started
[19:15:01][debug]: > request(storage(list(/ext/apps_manifests)))
[19:15:01][debug]: > request(storage(stat(/ext/Manifest)))
[19:15:01][debug]: > request(storage(read(/int/.region_data)))
[19:15:01][info]: connected
[19:15:01][debug]: > request(storage(info(/int)))
[19:15:01][info]: synchronize
[19:15:01][debug]: > request(system(setDate(2023-11-20 02:15:01 +0000)))
[19:15:01][debug]: < response(storage(stat(59233)))
[19:15:01][debug]: > request(system(property(devinfo.hardware.target)))
[19:15:01][error]: verify region: StorageError: File/Directory does not exist
[19:15:01][debug]: < response(storage(info(16 KiB / 76 KiB)))
[19:15:01][debug]: > request(storage(info(/ext)))
[19:15:02][debug]: < response(ok)
[19:15:02][info]: syncing date: (504s)
[19:15:02][debug]: > request(storage(stat(/ext/nfc/.mfkey32.log)))
[19:15:02][debug]: < response(system(property(hardware.target: 7)))
[19:15:02][debug]: < response(ok)
[19:15:02][debug]: > request(system(property(devinfo.firmware.api)))
[19:15:02][debug]: < response(storage(info(5.7 GiB / 119 GiB)))
[19:15:02][debug]: > request(system(property(devinfo.firmware.origin.fork)))
[19:15:02][debug]: < response(storage(stat(0)))
[19:15:02][debug]: > request(storage(read(/any/favorites.txt)))
[19:15:02][debug]: < response(system(property(firmware.api.major: 35)))
[19:15:02][debug]: < response(system(property(firmware.api.minor: 1)))
[19:15:02][debug]: < response(ok)
[19:15:02][debug]: < response(system(property(firmware.origin.fork: Xtreme)))
[19:15:02][debug]: > request(system(property(devinfo.firmware.origin.git)))
[19:15:02][debug]: < response(ok)
[19:15:03][debug]: < response(system(property(firmware.origin.git: )))
[19:15:03][debug]: > request(storage(write(/any/favorites.txt, 0 bytes)))
[19:15:03][debug]: < response(ok)
[19:15:03][debug]: < response(ok)
[19:15:03][info]: syncing archive: (861s)
[19:19:06][info]: disconnected
[19:19:06][debug]: reconnecting
[19:19:06][info]: connecting
[19:19:06][info]: session started
[19:19:09][info]: connected
[19:19:09][debug]: > request(storage(stat(/ext/Manifest)))
ts)))
[19:19:09][debug]: > request(storage(info(/int)))
[19:19:09][info]: synchronize
[19:19:09][debug]: > request(system(setDate(2023-11-20 02:19:09 +0000)))
[19:19:10][debug]: < response(storage(stat(59233)))
[19:19:10][error]: verify region: StorageError: File/Directory does not exist
[19:19:10][debug]: > request(system(property(devinfo.hardware.target)))
[19:19:10][debug]: < response(storage(info(16 KiB / 76 KiB)))
[19:19:10][debug]: > request(storage(info(/ext)))
[19:19:10][debug]: < response(ok)
[19:19:10][info]: syncing date: (398s)
[19:19:10][debug]: > request(storage(stat(/ext/nfc/.mfkey32.log)))
[19:19:10][debug]: < response(system(property(hardware.target: 7)))
[19:19:10][debug]: < response(ok)
[19:19:10][debug]: > request(system(property(devinfo.firmware.api)))
[19:19:10][debug]: < response(storage(info(5.7 GiB / 119 GiB)))
[19:19:10][debug]: > request(system(property(devinfo.firmware.origin.fork)))
[19:19:10][debug]: < response(storage(stat(0)))
[19:19:10][debug]: > request(storage(read(/any/favorites.txt)))
[19:19:10][debug]: < response(system(property(firmware.api.major: 35)))
[19:19:10][debug]: < response(system(property(firmware.api.minor: 1)))
[19:19:10][debug]: < response(ok)
[19:19:11][debug]: < response(system(property(firmware.origin.fork: Xtreme)))
[19:19:11][debug]: > request(system(property(devinfo.firmware.origin.git)))
[19:19:11][debug]: < response(ok)
[19:19:11][debug]: > request(storage(write(/any/favorites.txt, 0 bytes)))
[19:19:11][debug]: < response(system(property(firmware.origin.git: )))
[19:19:11][debug]: < response(ok)
[19:19:11][debug]: < response(ok)
[19:19:11][info]: syncing archive: (821s)
```

Willy-JL, 2023-11-28:
those are qflipper logs, they are useless. we need serial logs.

Willy-JL, 2023-12-09:
the nfc stack has been fully refactored. please check if it happens on the latest release @alphakdelta

alphakdelta, 2023-12-10:
I apologize for not getting those logs. The freezing no longer occurs and the flipper doesn't time out but now Amiibo(s) aren't registering at all on the Switch. I will do my best to upload some serial logs.

Willy-JL, 2023-12-13:
resolved on discord. for anyone wondering, something got messed up with the saved files. downloading the amiibo files again solved the issue. and the freezing was fixed via the refactor merged in v52, so im closing this issue as resolved

i know that some nfc files support being written to by other devices and flipper will save the original card as well as the new written data... i wonder if that is what happened, maybe the switch retaliated and overwrote the data? if so, the old non-working files should show a button like "revert to original" or something like that

closed 2023-12-13 by Willy-JL