Flipper-XFW/Xtreme-Firmware #461: Bad KB (Bluetooth) & BLE advertising combined?

analyserdmz, 2023-11-22:
I was wondering if it would be entirely impossible to execute a bad keyboard attack if a target device were to actually tap or click on the 'pair' button that appears when spoofing with BLE.

Just think about it. What if...?

Mr-Proxy-source, 2023-11-25:
I don't think that is possible.

pryh4ck, 2023-11-25:
I'm assuming I'm reading wrong and you're asking to execute a BLE HID attack WITHOUT pairing them first? No, however Bluetooth attacks get more and more sophisticated, so with kali or especially blackarch or if you spent some time researching anything is technically possible. Consider other ways to do HID attacks like mousejacking I'm not sure I haven't done looked in a while. Why Don't you get termux on your phone or if iPhone a kali vm (/UTM has iOS micro containers)

Willy-JL, 2023-11-28:
really depends on the implementation. might be possible, needs some tinkering. its on my radar, just not a priority.