Flipper-XFW/Xtreme-Firmware #470: SubGHz External Module - bug report + possible resolution

bip3r, 2023-11-28:
### Describe the bug.

I had an issue with external module C1011 for SubGHz.
With Xtreme FW the SubGHz app didn't see the module.
To test if the module itself acctually works, I installed Unleashed software.

But let's be fair - Xtreme soft is MUCH better, so I decided to revert to Xtreme despite the above "bug".
I was very surprised when after such a procedure SubGHz started to see the external module without a problem.

So it's acctually not a bug report, rather a hint to those who (for some unknown reason) have the same issue as I had.

### Reproduction

Attach C1011 module and check if in SubGhz -> Radio settings -> Module it shows External or just Internal without possibility to choose/change to External

Sil333033, 2023-11-28:
Open Xtreme app, press protocols, gpio pins and set SPI CC1101 HANDLE to DEFAULT.

Please lmk if its now still not detected

Willy-JL, 2023-11-28:
or maybe its justa a poor quality module, or it wasnt connected properly, either way user error

closed 2023-11-28 by Willy-JL