Flipper-XFW/Xtreme-Firmware #474: GPS from Mobile Phone

geckom, 2023-12-04:
### Description of the feature you're suggesting.

Since board are being created withe GPS now, and the GPS can be fairly flaky. It would seem like a better option to use GeoLocation built into mobile phones.

Using something like GPS2IP Lite (https://apps.apple.com/us/app/gps2ip-lite/id1562823492) you can get BluetoothLE to include "Location and Navigation" attributes on the property 0x2A67. This is pretty standardised and very easy to extract longitude and latitude from.

[image]
Shows the Notify property which can be subscribed to using something like:

```
        uint8_t notification[2] = {0x01, 0x00};
        uint16_t uuid = 0x2a67;
        aci_gatt_write_char_desc(gap->service.connection_handle, uuid, 2, (uint8_t *) &notification[0]);
```

Then simply ingesting the GPS data as it is received. I have attempted to change the firmware but my C coding skills are very rusty.

Willy-JL, 2023-12-09:
interesting idea... if the characteristic is included it might be readable and doable on flipper side, hmmm

ALEEF02, 2023-12-30:
I've been looking at this personally on https://github.com/ALEEF02/GPS_Exchanger_Flipper and I would love to see more support on this.

geckom, 2024-01-12:
For anyone wanting to wardrive with SubGhz, in the meantime I am using a mobile app which saves your location to a GPX record and then a quick python script to match it up with the .sub file's timestamp and injecting the lat/lon. More steps than it needs to be but is a reasonable bandaid solution.