Flipper-XFW/Xtreme-Firmware #482: Add NFC Keys to Dict

icemirr0r, 2023-12-09:
### Describe the enhancement you're suggesting.

Hi, please add the following keys to mf_classic_dict.nfc

#Misc. keys from hotels & library cards in Germany
914f57280ce3
324a82200018
370aee95cd69
2e032ad6850d
1feda39d38ec
288b7a34dbf8
0965e3193497
18c628493f7f
064d9423938a
995fd2a2351e
7c7d672bc62e
217250fb7014
ae7478ccaee7
abbf6d116eaf
05862c58edfb
e43b7f185460
6a59aa9a959b
b79e5b175227
7bc9ebb8274b
b2afbf2331d4
223e5847dd79
640524d2a39b
aee297cb2fd6
3da5dfa54604
0cf1a2aa1f8d

N-coder82, 2023-12-09:
Just use MFkey32

N-coder82, 2023-12-09:
It adds it to ur user duct

Sil333033, 2023-12-09:
Thanks for the recommendation. Have you checked these keys works on different cards?

icemirr0r, 2023-12-09:
> It adds it to ur user dict

I know, checked it with different cards and wanted to share/submit to the dict, not just my user dict.

> Thanks for the recommendation. Have you checked these keys works on different cards?
Yes with at least one additional card of the same location/facility.

296475202, 2024-01-09:
[extended-std.keys.txt](https://github.com/Flipper-XFW/Xtreme-Firmware/files/13881001/extended-std.keys.txt)
Key from Mifare Classic Tool

Willy-JL, 2024-01-13:
> [extended-std.keys.txt](https://github.com/Flipper-XFW/Xtreme-Firmware/files/13881001/extended-std.keys.txt) Key from Mifare Classic Tool

we already have those in the dictionary