Flipper-XFW/Xtreme-Firmware #486: Bluetooth link with official iOS flipperzero app doesn't work with Xtreme

wewlad-dev, 2023-12-10:
### Describe the bug.

Using the official firmware in flipperzero, I can pair my iOS flipperzero app through bluetooth with no issues, flashing Xtreme however, render that link broken, and I can't pair it anymore, does Xtreme supports using the iOS app? 

### Reproduction

1. Install Xtreme latest on flipper zero
2. Use the Official iOS flipper zero app (iOS 17.1.2)
3. Bluetooth link won't pair.

### Target

iOS 17.1.2

### Anything else?

Both Xtreme FW, flipperzero app, and iOS are the latest with all updates.

Willy-JL, 2023-12-10:
basic troubleshooting. unpair on both devices and pair again.

closed 2023-12-10 by Willy-JL

wewlad-dev, 2023-12-11:
That didn't work after flashing it from the official FW, I had to format the whole SD card and start fresh for it to work, so that part is solved if anyone got the same issue.
However, I'm opening this issue again as the Bluetooth connection is still buggy, if I pair flipperzero (loaded with  Xtrem FW) with iPhone app, and leave the flipperzero in standby mode for few hours, the iOS app won't detect the flipper bluetooth automatically, and the steps I had to do every time is delete the pairing profile within the app, pair again with temporarily password, and only then I can access the flipperzero, and I have to do this every time I leave flipper zero on standby for some time.

Willy-JL, 2023-12-11:
> on both devices

theres a reason for me saying things. wiping your sd card essentially wiped the bluetooth paired devices file, which clicking "forget all devices" in flipper settings would have done too. like it has done for everyone else with that issue, due to incompatible keys with our modified ble pairing

and that new issue you report does not sound right. have you tried closing and reopening the app? or rebooting flipper by holding back and left? or turning off and on bluetooth on the phone/flipper? maybe the iphone is somewhat disconnecting it due to inactivity