Flipper-XFW/Xtreme-Firmware #487: Flipperzero crashes and reboot after scanning with NFC HID iClass (picopass)

wewlad-dev, 2023-12-11:
### Describe the bug.

The flipper zero Xtreme FW crashes if I scan an NFC HID iClass (picopass) card, and when it prompts to save the scanned profile with the default name (name+date), it crashes and reboot, however, if I rename it to something else before saving, nothing happens, so it must be the name crashes it with memory buffer overflow or similar.

### Reproduction

1. Go to Apps
2. Got to NFC
3. Click "HID iClass (picopass)" and read card
4. once read, save the profile with the default name
5. Flipper reboot

### Target

XFW-0052

bmt626, 2023-12-12:
I am having the same issue I even deleted the apps folder and reflashed the latest stable update to make sure it was not something from the original updat flash.

Willy-JL, 2023-12-13:
i have pushed e06837de660b71ce2da9671c33832ac10d2b764c (update from apps submodule) which should hopefully resolve the issue. let me know if it causes other weird behavior

bettse, 2023-12-13:
> i was thinking that increasing the max name length for picopass devices would be a better solution.

I see no problem.  CC me on the PR and I'll 👍

closed 2024-02-02 by Willy-JL