Flipper-XFW/Xtreme-Firmware #488: Unlock SLIX-L--> Auth As TonieBox is missed in 0052

Bolicat, 2023-12-11:
### Description of the feature you're suggesting.

Dear Team XFW,

Firstly, I'd like to express my gratitude for your wonderful work – you've truly done an excellent job.

Now, I have a question regarding the current firmware XFW-0052:

Since the latest update, I've been missing an NFC function that I used to enjoy, specifically the pathway: NEF->Extra Actions--> Unlock SLIX-L--> Auth As TonieBox.

After the update, my previously configured NFC no longer functions as it did before. My question to you is: Is it possible to re-enable this function?

Thank you in advance for your assistance, and once again, I appreciate your outstanding work.

Best regards, 

bolicat

ClaraCrazy, 2023-12-11:
read the changelog. its gone bc ofw felt the need to refactor nfc and break it

will be back one day, possibly

Bolicat, 2023-12-12:
OK, I'll probably wait for the next version. Thanks for the quick response!

Hunduster, 2024-02-12:
Unfortunately still missing in 0053 :-(

Rugged-ESP, 2024-02-20:
Unfortunately it does not work.

brrn, 2024-03-11:
Working for me on DEV 11-03-2024

krisidmisso, 2024-06-26:
still missing in XFW-DEV 15-10-2023