Flipper-XFW/Xtreme-Firmware #494: Support spoofing remoteId ?

gled-rs, 2023-12-15:
### Description of the feature you're suggesting.

I wonder if it would be possible to support RemoteId broadcast, as required by the FAA when flying a registered drone in the US, but also in a lot of different countries.

An example project here: https://github.com/jjshoots/RemoteIDSpoofer/tree/master

The main issue with remoteID is that it broadcast not only the drone GPS Location, but also the operator, making it pretty unsafe to fly a drone in some areas. It may be safer to fly with a dozen or so reported locations around...

Willy-JL, 2023-12-15:
What frequency is it? And what kind of protocol?

gled-rs, 2023-12-16:
Still learning about it, but it seems to be WiFi or Bluetooth broadcast.  So probably Bluetooth, or we need a WiFi module.

FlashGibson, 2023-12-20:
> What frequency is it? And what kind of protocol?

Signed up to just to say this would be a fantastic idea. The FPV community would go wild for something like this. Joshua Bardwell has a good video on spoofing but it isn't very well polished. [Joshua Bardwell has a great video on it here](https://youtu.be/Q8jn_6EmYxE?si=Y7N4OCz-NxilbFI4). It might shine more light on how it actually works.

scott555, 2023-12-29:
This isn't a response to solve your request for the Xtreme firmware, but my workaround until something else is developed.  I use an app on my Android phone (I personally use an iphone, but I keep my previous Galaxy phone for all of my DJI uses).  It spoofs the GPS location to whatever you want it to be.   Its called FakeGPS and it's free, stable, and small.

Willy-JL, 2024-02-22:
ive had a very cursory look at the codebase, it seems to use both wifi and ble. from what i can tell, its ble for the actual remoteid stuff, and wifi for the web interface. although some chips dont even have ble, so im not sure. as for wifi, its purely adv data, so it would be doable with the new extra beacon api