Flipper-XFW/Xtreme-Firmware #495: Marauder Does Not Seem To Work

speedtoys, 2023-12-15:
### Describe the bug.

Just installed XFW-0052_09122023 using the downloaded Qflipper image, and this was after installing the same via WWW, just to make sure.

I have the ESP32, and in prev firmware it worked very well.

But when I start it and do an AP scan, it just sits there telling me to 'Press BACK to send stopscan' and no APs show up.

When I go to list APs, there are none to list.

The flipper installed code says 08122023, while the latest release says 09122023.   Just something I noticed.

### Reproduction

Pretty simple, application doesnt do anything, no significant steps other than opening and trying to use it.

Sil333033, 2023-12-15:
Hey.
Marauder is one of the most used apps, wouldn't it be stupid if we release with a broken marauder?

Unplug the board, if that doesn't fix it reflash it via the ESP Flasher in the GPIO folder..

ivanrylov, 2023-12-17:
I have the same issue. There is only "[MAYHEM] Marauder" (which is not expected to work with the ESP board, I think). There is no Marauder app for ESP.

Sil333033, 2023-12-17:
@ivanrylov Apps->WiFi->Marauder

N-coder82, 2023-12-17:
It's all the way at the bottom

ivanrylov, 2023-12-17:
> @ivanrylov Apps->WiFi->Marauder

Thanks! Sorry, I was looking in Apps -> GPIO.
Resolved.

Willy-JL, 2023-12-17:
just saw that the person saying its resolved is not OP

@speedtoys do this
1. Plug board to flipper
2. Apps > GPIO > ESP Flasher > Quick Flash > WiFi Dev> Marauder
3. When done, back out
4. Apps > WiFi > Marauder
5. Everything works as intended

closed 2023-12-17 by Willy-JL

jiriceKKK, 2023-12-25:
Hey guys I tried all of the advices flashed marauder even reinstalled it but it still seems to not scan anything. I'm currently using unleashed firmware if you know how to fix it let me know. Thank you all

Sil333033, 2023-12-25:
@jiriceKKK Marauder works fine. You are clearly doing something wrong.

Reflash your board. open the app in **APPS->WIFI->MARAUDER**

godmar, 2024-01-06:
I'm seeing the same issue as @speedtoys 

I flashed XFW-0052 08-12-2023 on the F0. On the Dev board, I flashed Marauder 13.6 via [fzeeflasher.github.io](https://fzeeflasher.github.io/), successfully based on the messages. I did this by connecting the board directly to the computer (without the F0)
```
---> Flashing firmware.
Image header, Magic=0xE9, FlashMode=0x02, FlashSizeFreq=0x2F
Writing data with filesize: 926048
Erase size 926048, blocks 57, block size 0x4000, offset 0x10000, encrypted no
Took 11772ms to write 926048 bytes
Erase size 0, blocks 0, block size 0x4000, offset 0x0000, encrypted no
---> Finished flashing firmware.

---> FLASHING PROCESS COMPLETED!
```

When I go to Apps->WiFi->`[ESP32] WiFi Marauder` and try scan, nothing happens.

When I try to flash from the F0 itself, Apps -> GPIO -> Quick Flash -> WiFi Dev and choose `Marauder (has Evil Portal)`, I get
[image]

```
Cannot connect to target. Error: 2
Make sure the device is in bootloader/reflash mode, then try again.
```

When I disconnect the board, hold the boot button, then reconnect, release the boot button, and try again, the same thing happens.

Can anyone describe possible diagnosis steps/different things to try?

godmar, 2024-01-06:
PS: I got it work as follows:

- power down the F0
- disconnect the wifi board
- disconnect the F0 from the computer
- reconnect the wifi board and power up the F0

Also, when everything is powered down, and I connect the wifi board to the F0, then connect via USB to the computer, it will also not work.  Disconnecting from the computer makes it start working, but the reconnecting it doesn't make it stop working.

Very odd.  Wish there was more diagnostic output.

ClaraCrazy, 2024-01-06:
you should ***NEVER*** have the wifi board connected to the PC and flipper at the same time when flashing it...

Willy-JL, 2024-01-06:
@godmar open esp flasher, then connect the board to flipper while holding the boot button. You need to use the quick flash option appropriate for your board, if you need to use the boot button, you most likely do not have an official WiFi dev board. Look at the chip, it most likely says "wroom". So use the "wroom" quick flash option. Then the apps in WiFi folder will work correctly.

godmar, 2024-01-06:
Thanks for the replies. I ordered the official WiFi dev board from the official site.  From what I can see, the chip reads `ESP32-S2-WRQVER` if I'm looking at the correct label:

[image]

> @godmar open esp flasher, then connect the board to flipper while holding the boot button. You need to use the quick flash option appropriate for your board,

When I tried this procedure while neither board nor F0 were connected via a USB cable, it worked. (Although it doesn't seem to have the latest Marauder version, it flashed v0.13.5 instead of v0.13.6.)

Willy-JL, 2024-01-06:
yeah that board supports automatic bootloader mode. your issue was being plugged in. and yes, marauder is updated along with the flipper firmware. use the devbuilds if you need something newer.

godmar, 2024-01-06:
> yeah that board supports automatic bootloader mode. your issue was being plugged in. and yes, marauder is updated along with the flipper firmware. use the devbuilds if you need something newer.

Just to clarify: the issue was the WiFi Dev board being plugged into the USB, correct?

It doesn't hurt to have the F0 plugged in and I can be using qFlipper, or do I need to unplug the F0 as well?

Willy-JL, 2024-01-06:
board was the issue

f0rg-02, 2024-03-27:
> just saw that the person saying its resolved is not OP
> 
> @speedtoys do this
> 
>     1. Plug board to flipper
> 
>     2. Apps > GPIO > ESP Flasher > Quick Flash > WiFi Dev> Marauder
> 
>     3. When done, back out
> 
>     4. Apps > WiFi > Marauder
> 
>     5. Everything works as intended

These are the correct steps that got my board to work. Yall are assholes in this thread.

Jake-perl, 2024-05-06:
@f0rg-02  It didn't work.