Flipper-XFW/Xtreme-Firmware #496: Something wrong with NFC 4k 7b reading

RekordNadoev, 2023-12-17:
### Describe the bug.

firmware: dev branch, commit 036e09fd3a4a01ce97d04c6f7b30abb0b745c830
Reading Mifare Classic 4k 7-byte UID with all the keys are in the dictionary returns different count of read sectors every time.
Something like
attempt 1: "Keys Found: 80/80, Sectors Read 16/40"
attempt 2: "Keys Found: 80/80, Sectors Read 38/40"
attempt 3: "Keys Found: 80/80, Sectors Read 24/40"
...
Every time sectors count differs.

(Unleashed firmware, dev branch last commit returns "Keys Found: 80/80, Sectors Read 40/40" every time.)

### Reproduction

1. Run: Menu => NFC => Read, with Mifare 4k 7-byte UID fob
2. Get message: Multi-protocol card, "Read as NTAG/Ultralight" or "Read as Mifare Classic"
3. Select "Read as Mifare Classic"
4. Get "Mifare Classic 4K, UID: 05 87 A6 8D D0 F1 00, Keys Found: 80/80, Sectors Read 33/40". Sometimes more or less sectors read.
5. Select: More => Save => enter filename => save
6. See "?? ?? ..." instead of some data in the dumps. 

### Anything else?

The fob is cheap one from aliexpress.
Unleashed firmware, dev branch, commit 364b33465416ea7f4bc1a71425fa5e4795c34083 returns "Keys Found: 80/80, Sectors Read 40/40" every time.

The part of saved dump:

Filetype: Flipper NFC device
Version: 4
# Device type can be ISO14443-3A, ISO14443-3B, ISO14443-4A, ISO14443-4B, ISO15693-3, FeliCa, NTAG/Ultralight, Mifare Classic, Mifare DESFire, SLIX, ST25TB
Device type: Mifare Classic
# UID is common for all formats
UID: 05 87 A6 8D D0 F1 00
# ISO14443-3A specific data
ATQA: 00 42
SAK: 18
# Mifare Classic specific data
Mifare Classic type: 4K
Data format version: 2
# Mifare Classic blocks, '??' means unknown data
Block 0: 04 12 19 C3 21 93 16 98 42 00 E3 20 00 00 00 00
Block 1: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 2: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 3: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 4: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 5: 22 31 63 F9 58 99 92 52 A8 00 2C B8 01 7C E6 E9
Block 6: 22 31 63 F9 58 99 92 52 A8 00 2C B8 01 7C E6 E9
Block 7: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF

Willy-JL, 2023-12-17:
Is that dump a successful read?

RekordNadoev, 2023-12-17:
There is the full dump made by Unleashed FW:
```
Filetype: Flipper NFC device
Version: 4
# Device type can be ISO14443-3A, ISO14443-3B, ISO14443-4A, ISO14443-4B, ISO15693-3, FeliCa, NTAG/Ultralight, Mifare Classic, Mifare DESFire, SLIX, ST25TB
Device type: Mifare Classic
# UID is common for all formats
UID: 05 87 A6 8D D0 F1 00
# ISO14443-3A specific data
ATQA: 00 42
SAK: 18
# Mifare Classic specific data
Mifare Classic type: 4K
Data format version: 2
# Mifare Classic blocks, '??' means unknown data
Block 0: 04 12 19 C3 21 93 16 98 42 00 E3 20 00 00 00 00
Block 1: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 2: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 3: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 4: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 5: 22 31 63 F9 58 99 92 52 A8 00 2C B8 01 7C E6 E9
Block 6: 22 31 63 F9 58 99 92 52 A8 00 2C B8 01 7C E6 E9
Block 7: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 8: 31 99 03 27 30 07 77 40 87 90 05 95 AB 0D 1D 86
Block 9: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 11: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 12: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 13: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 14: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
Block 15: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 16: 42 12 BB 41 E3 38 82 32 4B 00 00 00 00 00 DF 11
Block 17: 26 61 64 01 58 91 92 5C 80 03 7F 20 10 8C A9 42
Block 18: 26 61 64 01 58 91 92 5C 80 03 7F 20 10 8C A9 42
Block 19: 73 06 8F 11 8C 13 FF 07 80 00 2B 7F 32 53 FA C5
Block 20: 01 02 60 64 41 92 6F 00 00 00 00 00 00 00 00 00
Block 21: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 22: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 23: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 24: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 25: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 26: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 27: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 28: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 29: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 31: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 32: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 33: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 34: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 35: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 36: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 37: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 38: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 39: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 41: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 42: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 43: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 44: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 45: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 46: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 47: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 48: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 49: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 51: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 52: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 53: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 54: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[... 177 lines cut ...]
Block 232: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 233: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 234: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 235: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 236: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 237: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 238: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 239: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 240: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 241: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 242: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 243: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 244: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 245: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 246: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 247: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 248: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 249: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 250: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 251: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 252: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 253: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 254: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 255: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
```

RekordNadoev, 2023-12-17:
There is one of  dumps made by Xtreme FW:

```
Filetype: Flipper NFC device
Version: 4
# Device type can be ISO14443-3A, ISO14443-3B, ISO14443-4A, ISO14443-4B, ISO15693-3, FeliCa, NTAG/Ultralight, Mifare Classic, Mifare DESFire, SLIX, ST25TB
Device type: Mifare Classic
# UID is common for all formats
UID: 05 87 A6 8D D0 F1 00
# ISO14443-3A specific data
ATQA: 00 42
SAK: 18
# Mifare Classic specific data
Mifare Classic type: 4K
Data format version: 2
# Mifare Classic blocks, '??' means unknown data
Block 0: 04 12 19 C3 21 93 16 98 42 00 E3 20 00 00 00 00
Block 1: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 2: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 3: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 4: ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ??
Block 5: 22 31 63 F9 58 99 92 52 A8 00 2C B8 01 7C E6 E9
Block 6: 22 31 63 F9 58 99 92 52 A8 00 2C B8 01 7C E6 E9
Block 7: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 8: ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ??
Block 9: ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ??
Block 10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 11: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 12: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 13: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 14: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
Block 15: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 16: 42 12 BB 41 E3 38 82 32 4B 00 00 00 00 00 DF 11
Block 17: 26 61 64 01 58 91 92 5C 80 03 7F 20 10 8C A9 42
Block 18: 26 61 64 01 58 91 92 5C 80 03 7F 20 10 8C A9 42
Block 19: 73 06 8F 11 8C 13 FF 07 80 00 2B 7F 32 53 FA C5
Block 20: 01 02 60 64 41 92 6F 00 00 00 00 00 00 00 00 00
Block 21: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 22: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 23: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 24: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 25: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 26: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 27: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 28: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 29: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 31: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 32: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 33: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 34: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 35: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 36: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 37: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 38: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 39: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 41: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 42: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 43: FF FF FF FF FF FF ?? ?? ?? ?? FF FF FF FF FF FF
Block 44: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 45: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 46: ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ??
Block 47: FF FF FF FF FF FF ?? ?? ?? ?? FF FF FF FF FF FF
Block 48: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 49: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 51: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 52: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 53: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[... 178 lines cut ...]
Block 232: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 233: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 234: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 235: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 236: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 237: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 238: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 239: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 240: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 241: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 242: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 243: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 244: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 245: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 246: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 247: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 248: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 249: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 250: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 251: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 252: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 253: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 254: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 255: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
```

RekordNadoev, 2023-12-18:
Today I changed  FW from Xtreme to Unleashed and back to Xtreme for some testing.
Now the fob reading is 40/40 sectors every time.
It seems like it was some kind of glitch.
Sorry.

RekordNadoev, 2023-12-18:
It seems like it was some kind of glitch.

closed 2023-12-18 by RekordNadoev