Flipper-XFW/Xtreme-Firmware #498: Philips Sonicare does not working

lordrak007, 2023-12-17:
### Describe the bug.

Whan i read brush it detect unknown chip and wants to unlock by password.

### Reproduction

Read brush head by NFC

Sil333033, 2023-12-17:
Can you send a pic of what its detecting?

lordrak007, 2023-12-17:
[image]
Can i help with debug?

Sil333033, 2023-12-17:
yes. Can you send the whole dump here of the file?

lordrak007, 2023-12-17:
[image]

lordrak007, 2023-12-17:
[NTAG213_2023-12-17_23,28.zip](https://github.com/Flipper-XFW/Xtreme-Firmware/files/13697987/NTAG213_2023-12-17_23.28.zip)

Sil333033, 2023-12-21:
thanks, just looked at the dump and it looks like it missing some data we are filtering on to verify it is a Sonicare brush.

page 5 byte 4 until page 8 byte 2 is 70 68 69 6C 69 70 73 2E 63 6F 6D. if you convert this to ascii you get **philips.com**, we are looking for philips.com**/nfcbrushheadtap** in the code.. So thats the issue, i will fix this later tonight

lordrak007, 2024-02-07:
Can i somehow participate to improve detection?