Flipper-XFW/Xtreme-Firmware #499: Selecting new html file for evil portal results in overlapping portals

FlashGibson, 2023-12-20:
### Describe the bug.

Selecting new html file for evil portal results in overlapping portals. 

### Reproduction

1. Start Evil Portal
2. Stop Evil Portal
3. Select new HTML file
4. Start Evil Portal
5. Connect using new device
6. Captive portal shows multiple, overlapping portals with random numbers

### Target

Evil Portal

### Anything else?

[Using bigbrodude6119's repo for html files](https://github.com/bigbrodude6119/flipper-zero-evil-portal
[image]
)

Willy-JL, 2024-02-13:
from what i remember, this was a bug in the evil portal esp firmware. this is not used anymore, now marauder has evil portal built in, without this bug. please update your flipper, flash the board using esp flasher quick flash, and try again.

suigeneris58, 2024-04-07:
Hello @FlashGibson could you fix this issue? I have the same problem.
Regards

hxcking, 2024-04-09:
> Hello @FlashGibson could you fix this issue? I have the same problem. Regards

To fix it you can delete the index.html file from the directory of Evil Portal (data_apps > evilportal)  and upload your html files in data_apps>marauder>html folder.