Flipper-XFW/Xtreme-Firmware #503: NFC - Failed identifying credit cards on latest XFW-0052_09122023

OscarAkaElvis, 2023-12-30:
### Describe the bug.

On the latest firmware XFW-0052_09122023, the credit card detection by NFC is not working well. It was working on the older XFW-0051_01092023 firmware version.

This is the result of reading a credit card on this latest release (XFW-0052_09122023). It is unknown as you can see:
[image]

This is the same credit card but using the September firmware release (XFW-0051_0109202):
[image]

### Reproduction

1. Switch on
2. Enter on NFC menu
3. Launch the "Read" option while holding a credit card
4. Check the results.

Sil333033, 2023-12-30:
i don't understand why you updated your flipper without reading the changelog🦐

closed 2023-12-30 by Sil333033

OscarAkaElvis, 2023-12-30:
I did quickly. Maybe I didn't notice that. Now I see it. Looking forward for the change. Sorry for the inconvenience.