Flipper-XFW/Xtreme-Firmware #504: nfc writing and partially reading wont work

Jona-Walpert, 2024-01-01:
### Describe the bug.

## Issue with NFC Tag Reading and Writing on Flipper Zero with Xtreme Firmware

### Problem Description

I encountered an issue with my Flipper Zero running the Xtreme Firmware, where I am unable to read or write NFC tags successfully. When attempting to read a new and empty tag, the device reports that only 133 out of 135 pages were read. However, I can successfully read all pages if the tags are unlocked using the password FF FF FF FF wich I found online. 

Additionally, when trying to write to an NFC tag, the device indicates that it is still locked, and there is an issue related to Auth 0 (refer to the attached picture).

### Steps to Reproduce

1. Attempt to read a new and empty NFC tag.
2. Observe the reported read result (133/135 pages).
3. Try reading the same tag after unlocking with the password FF FF FF FF.
4. Attempt to write to an NFC tag.
5. Observe the reported lock status and Auth 0 issue (as shown in the attached picture).

### Expected Behavior

The Flipper Zero with Xtreme Firmware should be able to read and write to NFC tags without encountering issues related to incomplete reads or authentication problems.

### Additional Information

- Flipper Zero model: [FZ.1]
- Xtreme Firmware version: [XFW-0052]
- Any relevant screenshots or error messages: [see picture]

[image]

### Reproduction

1. Switch on
2. nfc
3. saved
4. select saved nfc file
5. write
6. error

### Logs

```Text
910 [APP] qFlipper version 1.3.3 commit bfce851d 2023-11-10T10:40:31
910 [APP] OS info: Windows 10 Version 22H2 10 10.0.19045 Qt 6.4.2
924 [DBG] Trying serial port  at \\.\COM1
924 [DBG] Trying serial port FLIP_FLIPPER at \\.\COM10
924 [DBG] Using  serial port FLIP_FLIPPER at \\.\COM10
925 [RPC] Starting RPC session...
1030 [RPC] RPC session started successfully.
1031 [RPC] (1) System Protobuf Version START
1038 [UPD] Fetched update information from https://update.flipperzero.one/qFlipper/directory.json
1105 [DBG] Detected protobuf version: 0.21
1105 [RPC] (1) System Protobuf Version SUCCESS
1106 [RPC] (2) Property Get START
1139 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
1239 [RPC] (2) Property Get SUCCESS
1239 [RPC] (3) Storage Info @/ext START
1291 [RPC] (3) Storage Info @/ext SUCCESS
1292 [RPC] (4) Storage Stat @/ext/Manifest START
1357 [RPC] (4) Storage Stat @/ext/Manifest SUCCESS
1357 [RPC] (5) System Get DateTime START
1422 [DBG] Flipper time skew is -1431 milliseconds
1422 [RPC] (5) System Get DateTime SUCCESS
1422 [RPC] (6) System Set DateTime START
1475 [RPC] (6) System Set DateTime SUCCESS
1475 [RPC] Stopping RPC session...
1477 [RPC] Attempting to unload protobuf plugin...
1477 [RPC] Unloaded protobuf plugin.
1477 [RPC] RPC session stopped successfully.
1477 [REG] Registering the device
[... 232 lines cut ...]
46540 [RPC] (115) Gui Send Input START
46592 [RPC] (115) Gui Send Input SUCCESS
46760 [RPC] (116) Gui Send Input START
46823 [RPC] (116) Gui Send Input SUCCESS
46827 [RPC] (117) Gui Send Input START
46892 [RPC] (117) Gui Send Input SUCCESS
46892 [RPC] (118) Gui Send Input START
46957 [RPC] (118) Gui Send Input SUCCESS
49693 [RPC] (119) Gui Send Input START
49757 [RPC] (119) Gui Send Input SUCCESS
49776 [RPC] (120) Gui Send Input START
49828 [RPC] (120) Gui Send Input SUCCESS
49828 [RPC] (121) Gui Send Input START
49890 [RPC] (121) Gui Send Input SUCCESS
51428 [RPC] (122) Gui Send Input START
51492 [RPC] (122) Gui Send Input SUCCESS
51492 [RPC] (123) Gui Send Input START
51559 [RPC] (123) Gui Send Input SUCCESS
51559 [RPC] (124) Gui Send Input START
51623 [RPC] (124) Gui Send Input SUCCESS
```

### Anything else?

Logs always says succes. writing or readng all pages didnt work but logs show SUCCESS

Jona-Walpert, 2024-01-01:
[image]

Sil333033, 2024-01-01:
your card is locked by the lock bits. even with the passwords for all the pages you cant do anything. 
Those logs are useless btw, ignore them. They are referring to your actions via qflipper

Jona-Walpert, 2024-01-01:
> your card is locked by the lock bits. even with the passwords for all the pages you cant do anything. Those logs are useless btw, ignore them. They are referring to your actions via qflipper

Ok but I should be able to write it?

Jona-Walpert, 2024-01-01:
> words for all the pages you c

Because I still  cant write to a nfc tag... So the Issue is still present

Willy-JL, 2024-01-01:
explain how you are attempting to write to it.

Jona-Walpert, 2024-01-01:
I go to "NFC". then to "saved". Then I choose a saved nfc file. After that I click on "Write" and hold a nfc tag to the back of the flipper and Then the message apepars that you can see in the first picture

Willy-JL, 2024-01-01:
learning to use google might prove fruitful.
https://www.google.com/search?q=write+locked+ntag215+with+flipper+zer
https://www.reddit.com/r/flipperzero/comments/18e48gg/flipper_thinks_all_ntag215_cards_are_password/
https://github.com/flipperdevices/flipperzero-firmware/issues/3281
https://github.com/flipperdevices/flipperzero-firmware/pull/3285

the issue is resolved and merged into the firmware. it is already available in dev builds, and will be released in v53.

Willy-JL, 2024-01-01:
technically not resolved in a stable release however, so i will reopen and mark as pending release.

closed 2024-02-02 by Willy-JL