Flipper-XFW/Xtreme-Firmware #505: cant write to nfc tag

Jona-Walpert, 2024-01-01:
your card is locked by the lock bits. even with the passwords for all the pages you cant do anything. 
Those logs are useless btw, ignore them. They are referring to your actions via qflipper

_Originally posted by @Sil333033 in https://github.com/Flipper-XFW/Xtreme-Firmware/issues/504#issuecomment-1873296156_

Willy-JL, 2024-01-01:
do not open a second issue.

closed 2024-01-01 by Willy-JL

Jona-Walpert, 2024-01-01:
> do not open a second issue.

can a Issue be opened again after closed?

Willy-JL, 2024-01-01:
if it has good reason to.

Sil333033, 2024-01-01:
can a Issue be opened again after closed?

Jona-Walpert, 2024-01-01:
> if it has good reason to.

I mean I still cant write to a nfc tag. the reading problem was just a misunderstanding

Willy-JL, 2024-01-01:
stop talking here dude. take a goddamn hint.