Flipper-XFW/Xtreme-Firmware #506: New Features for frequency analyzer.

X3r0C00L, 2024-01-02:
### Describe the enhancement you're suggesting.

I would like to request the ability to mute certain frequencies during analyzing. Reason being is there are a few times when I want to see what is out there but find that there are usually a couple of transmitters nearby that are constantly broadcasting, which overshadows all other frequencies.  As an example, Ill ge outside my house scanning to see if there are any weather stations broadcasting or something similar, but there is a tower close by that broadcasts in the 300mhz range constant and it keeps me from being able to see other frequencies because it stays stuck on that one. 
Also would like to see a way to manually input a range to scan. like if I only wanted to scan from 430-440mhz (I know we have spectrum anaylizer app) but this feature would be nice in the freq analyzer. 

### Anything else?

Great job on everything. Ive been using Xtreme since day one. Thank you all so much.

sadi-code, 2024-02-14:
Faça um filtro passa faixa fisicamente.  Use componentes físicos LCR também ajuste sua antena para o comprimento de onda que desejas. Assim você elimina sinais indesejáveis. Para um software funcionar perfeitamente,  você deve trabalhar com o hardware compatível e próprio para tal função.

X3r0C00L, 2024-02-21:
Não é viável para nós um tanque LC ou filtro passa-banda para minhas necessidades. Estou querendo analisar para ver se há frequências transmitindo na faixa 430-450 mas tenho dois transmissores perto de mim que transmitem em 433.545 e 451.750 que ofuscam completamente todo o resto. Este é apenas um caso, pois há outras faixas de frequência que gostaria de analisar que apresentam problemas semelhantes. Neste momento estou tendo que usar meu HackRF one Portapack H2 para encontrar o que procuro.
Obrigado pela orientação.